esp-cpp/espp #97: BLDC Driver issue

guo-max, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What is the expected behavior?

I am review the bldc_driver code by comparing to your code and found some issue. 

`bldc_driver.set_pwm(25,25,25);` // for example
I assume the A,B,C high pin should be 25% high, 75% low. But instead, the high pin now is 75% high, 25% low. 

`    ESP_ERROR_CHECK(mcpwm_generator_set_actions_on_compare_event(
        gen_high,
        MCPWM_GEN_COMPARE_EVENT_ACTION(MCPWM_TIMER_DIRECTION_UP, comp, MCPWM_GEN_ACTION_HIGH),
        MCPWM_GEN_COMPARE_EVENT_ACTION(MCPWM_TIMER_DIRECTION_DOWN, comp, MCPWM_GEN_ACTION_LOW),
        MCPWM_GEN_COMPARE_EVENT_ACTION_END()));
  }
`
This function set gpio `high` at `up direction` and `low` at `down direction`. 
For a counter counting from 0 -> 100->0, if set to 25%, the gpio will set `high` at `24->25` and set `low` at `26->25`, which will give `high` time from `25->100->25` which is 75% high. 

### What is the actual behavior?

When set pwm to 25%, the high side at a_high_gpio show be 25% high, 75% low. 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

I know it works with your FOC controller. If this is a issue, what will change in the FOC controller?
All the highs lows are inverted, May only change the motor direction?

finger563, 2023-07-31:
@guo-max It definitely doesn't just change the motor direction - as an example, I've changed the `GEN_ACTION` for the code in question to invert it (so `UP` -> `LOW` and `DOWN` -> `HIGH`), and simply inverting the direction of the motor does not seem to fix it, but the motor stalls when driving output. I'll dig into it some more to see what the cause may be. another option is to simply inver the input provided (so that inputting 25 results in 75 output) but that seems like a kludge to me.

finger563, 2023-08-01:
You're correct - if you keep everything else the same (including the `sensor_direction`, then there is no discernable difference :) I'll push the fix 👍

PR #100 (2023-08-01): fix(bldc_driver): fix driver level inversion

## Description

* Update to set output to low when counting up, and high when counting down. This means that setting a value of 25% will actually have the output high for 25% of the time.
* Update examples to always calibrate by setting zero electrical offset to 0.

## Motivation and Context

Fixes #97

⚠️ You will have to recalibrate your motors after upgrading to this code ⚠️ 

## How has this been tested?

Building and running the bldc motor and bldc haptics examples.

## Screenshots (if appropriate, e.g. schematic, board, console logs, lab pict [... cut]
commit 4b814f04 (2023-08-01): fix(bldc_driver): fix driver level inversion (#100)

* fix(bldc_driver): fix driver level inversion
* Update to set output to low when counting up, and high when counting down. This means that setting a value of 25% will actually have the output high for 25% of the time.
Fixes #97

* feat(bldc): updated examples
* Update examples to always calibrate by setting zero electrical offset to 0.

* example(bldc_motor): update
* Update bldc motor example to 1) always calibrate sensor direction, 2) always calibrate electrical angle, and 3) enable the motor so it actually runs

* fix(bldc_driver): confi [... cut]
closed 2023-08-01 by finger563