esp-cpp/espp #281: Change repo submodules from SSH to HTTPS

samiralavi, 2024-07-07:
### Answers checklist.

- [X] I have read the documentation and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Gitlab CI basically doesn't work when submodules are linked via SSH URL. For git repos that are hosted on other Git servers than Gitlab, only HTTP(S) is supported. I request reverting the recent change (https://github.com/esp-cpp/espp/pull/256) of linking submodules via SSH to HTTPS. Thanks.

References: 
- https://docs.gitlab.com/ee/ci/git_submodules.html#use-git-submodules-in-cicd-jobs
- https://www.perforce.com/manuals/gitswarm-ee/ci/git_submodules.html#:~:text=because%20GitLab%20CI%20only%20uses%20HTTP(S)%20for%20cloning%20your%20sources

finger563, 2024-07-07:
@samiralavi thanks for opening this issue, sorry it doesn't seem to be working for you in your environment. 

I've not used gitlab in a little while, does setting the variable `GIT_SUBMODULE_FORCE_HTTPS` not work when the git servers are hosted at GitHub? It seems that should force convert submodule references to https even if they were ssh to begin with. If that doesn't work, let me know and I'll look into switching these to https.

samiralavi, 2024-07-07:
Thanks @finger563 for the quick reply. That option only works if the submodule is hosted on Gitlab.

GitLab doesn't support built-in SSH key management (https://docs.gitlab.com/ee/ci/ssh_keys/). It is up to the developers to follow those steps in the link to add ssh keys to CI jobs, which means extra work. I was able to fix the issue by injecting SSH keys into the runners and my Github account, so keeping the submodules as is now is fine. I think if there is no specific requirement from you to use SSH links, HTTPS provides a more developer-friendly experience.

finger563, 2024-07-07:
Sounds good, glad there's a workaround till we shift over. I agree since there is nothing private we don't need ssh - I'll make a branch and switch over to make sure some of the places im using it as a submodule all work properly (they should). Expect a PR within a day or so

samiralavi, 2024-07-07:
Thanks. Btw, very useful components are here, great work.

PR #283 (2024-07-08): fix #281 switch from ssh to https submodules

## Description

Switch from ssh to https for submodules

## Motivation and Context

Closes #281 

In short, allows people who have not set up ssh / are using github to clone / fork / use this as a submodule.

## How has this been tested?

Clone, checkout this branch, submodule update
commit bc8100c0 (2024-07-08): fix #281 switch from ssh to https submodules (#283)
closed 2024-07-08 by finger563