esp-cpp/espp #310: Improve documemntation

JaviVilarroig, 2024-08-17:
### Is your feature request related to a problem?

I have been unable to find a clear explanation of how to deploy the espp component in my idf development environment.

Has this library been added to the Espressif Compnent register? I have been unable to find it.

It will be nice if the documentation starts with a little chapter explaining how to deploy the library.

Thanks!

finger563, 2024-08-17:
Hey @JaviVilarroig thanks for opening this issue - updating the documentation has def. been on my todo-list, so thanks for poking me about it 😅 

I'll make some updates to the documentation shortly, but really quick to answer your question and provide some help until then:

You have a couple of options:
1. You can use the [esp-cppp/template](https://github.com/esp-cpp/template) repository as a starting point, which is very similar to the esp-idf template, but is geared towards c++ development and already has `espp` as a submodule, along with the appropriate configuration in the top-level CMakeLists.txt.
2. If you have an existing project with a `components` directory, then you can clone `espp` as a submodule within that directory e.g. `git submodule add https://github.com/esp-cpp/espp components/espp`, then make sure to run `git submodule update --init --recursive`. Afterwards, simply update your `CMakeLists.txt` to add
  ```cmake
  # add the component directories that we want to use
  set(EXTRA_COMPONENT_DIRS
    "components/espp/components"
  )
  ```
3. You can clone espp somewhere on your computer and then point your project to its `components` directory to use any of the components it contains, similar to the step above.

The mention about the idf component registry is a good one - I don't personally use it much (preferring to use git submodules instead), but I'll add it as a todo to publish this as an idf-component so that its easier for others who use the registry to integrate into their projects.

JaviVilarroig, 2024-08-18:
Hi @finger563 

Thanks for the fast answer :)

Any of the three answers looks OK, thanks :) I will try when I have a minute.

I really think is good to try the component registry, because it has a good integration with the idf environment and most newbies in the ESP32 world, like me, will go there first.

I my case I was already starting to develop my own library because I didn't found anything similar in the registry. I found this one when I was checking if the name espp was already taken in a search engine. :smile: 

Thanks again!

PR #313 (2024-08-19): Update main readme

## Description

* Add additional information to the readme

## Motivation and Context

closes #310
commit 08857a4e (2024-08-19): Update main readme (#313)

* Add additional information to the readme
closes #310
closed 2024-08-19 by finger563